jgaeddert/liquid-dsp #290: Why is FM modulation factor limited to [0, 1]?

shamrin, 2022-06-26:
Currently `freqdem` limits *modulation factor* `_kf` to [0, 1]:

https://github.com/jgaeddert/liquid-dsp/blob/9423d76cec21e18c1dab55b5cc7d4c40ef8a4bc2/src/modem/src/freqdem.proto.c#L46-L48

In comparison, LuaRadio uses 1.25 as the default value for wide band FM demodulation:

https://github.com/vsergeev/luaradio/blob/ff82019f88098ff2f0f5d3786535b492070c64f8/radio/composites/wbfmmonodemodulator.lua#L28
https://luaradio.io/docs/reference-manual.html#frequencydiscriminatorblock

(LuaRadio calls it *modulation index*, but looking at the formula it's the same thing as liquid-dsp's *modulation factor*.)

Who is in the wrong here, and does it even matter?

jgaeddert, 2022-08-24:
It should be allowed to exceed 1; I think I was expecting this number to be normalized, but this doesn't preclude it from being larger than 1.

jgaeddert, 2022-08-24:
Relaxed constraint in 21b6db62

closed 2022-08-25 by jgaeddert