jgaeddert/liquid-dsp #292: FFTW

jrmims, 2022-08-11:
I've compiled liquiddsp but am not seeing evidence of FFTW being used in-place of the native liquid FFT functionality. ./configure states it sees FFTW on my system. Is there anything else I need to do to use FFTW's FFT capabilities? I ran liquid's FFT and FFTW in the same application to compare them and am certain liquid is not using FFTW based on speed. Thanks!

seonlee2013, 2022-08-11:
好啊，邮件收到啦！
Hi, your email is arrived safely

jgaeddert, 2022-08-24:
@jrmims What functions are you expecting FFTW to be used? `configure` should define the following in `config.h`:

    #define HAVE_FFTW3_H 1    
    #define HAVE_LIBFFTW3F 1    

This in turn specifies the use of FFTW in the [internal header](https://github.com/jgaeddert/liquid-dsp/blob/master/include/liquid.internal.h#L743) file to define macros used throughout the library.

jrmims, 2022-08-24:
I was expecting liquid to use fftw_plan_dft_1d instead of fft_create_plan (internally), for example. I made an application that ran Chrono for liquids fft and compared to fftw and the time duration was much larger with liquids. My expectations could be off or my install could be botched.

jgaeddert, 2022-09-25:
Just to follow up on this, running `./configure.sh` should indicate that it's found both the FFTW header(s) and library, in which case it will use FFTW internally for most functions that need a transform. You can confirm this by looking at `config.h` as indicated above. That's step 1.

If there are _specific_ functions that you are noticing are slow, please indicate to me which ones these are and I can check to see if they are using the internal FFT with liquid.

jrmims, 2022-10-05:
I am seeing fftw3 when I run configure.
"
Checking for fftw3.h usability... Yes
Checking for fftw3.h presence... Yes
Checking for fftw3.h... yes
Checking for fftwf_plan_dft_1d in -lfftw3f...  Yes
"

I do not see FFTW mentioned in /use/local/include/liquid.h

Slow down is in fft_execute, fft_create_plan. Vs fftw3's fftw_plan_dft_1d
and fftw_execute.

jrmims, 2022-10-14:
./configure is showing FFTW3 found.
config.h also shows the values above.

the functions I am comparing in speed are:
fft_create_plan vs fftw_plan_dft_1d
fft_execute vs fftw_execute
fft_destroy_plan vs fftw_destroy_plan

sample code:
`
int main(int argc, char *argv[]) {
    if(argc > 3) {
        std::string inputFile = argv[1];
        std::string outputFile1 = argv[2];
        std::string outputFile2 = argv[3];

        // Extract filename
        std::cout << "Attempting to open file. : " << inputFile << std::endl;
        // Open file
        std::ifstream fin(inputFile, std::ios::binary);
        if(!fin) {
            std::cout << "Could not open input file" << std::endl;
            return 1;
        }
        fin.seekg(0, std::ios::end);
        const size_t numInputElements = fin.tellg() / sizeof(std::complex<float>);
        fin.seekg(0, std::ios::beg);

        std::ofstream fout1(outputFile1, std::ios::binary);
        std::ofstream fout2(outputFile2, std::ios::binary);

        size_t bufferSize = 1000000;  // number of symbols per grab
        unsigned int fftSize = nextPowerOfTwo(bufferSize);  // value is 1048576

        // Grab data
        std::vector<std::complex<float>> inputSymbols(bufferSize);
        fin.read(reinterpret_cast<char*>(&inputSymbols[0]), bufferSize*sizeof(std::complex<float>));
        std::cout << "fftSize : " << fftSize << std::endl;
        // FFTW
        {
            std::vector<std::complex<double>> signalVector(fftSize, 0);
            std::copy(inputSymbols.begin(), inputSymbols.end(), signalVector.begin());
            std::vector<std::complex<double>> fftVector(fftSize, 0);
            auto start = std::chrono::high_resolution_clock::now();
            fftw_plan plan;
            plan = fftw_plan_dft_1d(fftSize, reinterpret_cast<fftw_complex*>(&signalVector[0]), reinterpret_cast<fftw_complex*>(&fftVector[0]), FFTW_FORWARD, FFTW_ESTIMATE);
            auto end1 = std::chrono::high_resolution_clock::now();
            fftw_execute(plan);
            auto end2 = std::chrono::high_resolution_clock::now();
            fftw_destroy_plan(plan);
            auto end3 = std::chrono::high_resolution_clock::now();
            auto total = std::chrono::duration_cast<std::chrono::microseconds>(end3 - start);
            auto dur1 = std::chrono::duration_cast<std::chrono::microseconds>(end1 - start);
            auto dur2 = std::chrono::duration_cast<std::chrono::microseconds>(end2 - end1);
            auto dur3 = std::chrono::duration_cast<std::chrono::microseconds>(end3 - end2);
            std::cout << "Total Time fftw: " << total.count() << " microseconds" << std::endl;
            std::cout << "create plan fftw: " << dur1.count() << " microseconds" << std::endl;
            std::cout << "execute fftw: " << dur2.count() << " microseconds" << std::endl;
            std::cout << "destroy fftw: " << dur3.count() << " microseconds" << std::endl;

            std::vector<std::complex<float>> outVector(fftVector.size());
            std::copy(fftVector.begin(), fftVector.end(), outVector.begin());
            // Output to file
            {
                for(size_t idx = 0; idx < fftVector.size(); idx++) {
                    fout1.write(reinterpret_cast<const char*>(&outVector[idx]), sizeof(std::complex<float>));
                }
            }
        }

        // Liquid DSP
        {
            std::vector<std::complex<float>> signalVector(fftSize, 0);
            std::copy(inputSymbols.begin(), inputSymbols.end(), signalVector.begin());
            std::vector<std::complex<float>> fftVector(fftSize, 0);
            auto start = std::chrono::high_resolution_clock::now();
            fftplan plan = fft_create_plan(fftSize, &signalVector[0], &fftVector[0], LIQUID_FFT_FORWARD, 0);
            auto end1 = std::chrono::high_resolution_clock::now();
            fft_execute(plan);
            auto end2 = std::chrono::high_resolution_clock::now();
            fft_destroy_plan(plan);
            auto end3 = std::chrono::high_resolution_clock::now();
            auto total = std::chrono::duration_cast<std::chrono::microseconds>(end3 - start);
            auto dur1 = std::chrono::duration_cast<std::chrono::microseconds>(end1 - start);
            auto dur2 = std::chrono::duration_cast<std::chrono::microseconds>(end2 - end1);
            auto dur3 = std::chrono::duration_cast<std::chrono::microseconds>(end3 - end2);
            std::cout << "Total Time Liquid: " << total.count() << " microseconds" << std::endl;
            std::cout << "create plan Liquid: " << dur1.count() << " microseconds" << std::endl;
            std::cout << "execute Liquid: " << dur2.count() << " microseconds" << std::endl;
            std::cout << "destroy Liquid: " << dur3.count() << " microseconds" << std::endl;

            // Output to file
            {
                for(size_t idx = 0; idx < fftVector.size(); idx++) {
                    fout2.write(reinterpret_cast<const char*>(&fftVector[idx]), sizeof(std::complex<float>));
                }
            }
        }     
        fin.close();
        fout1.close();
        fout2.close();
    } else {
        std::cout << "./ffttest <input data> <output file fftw> <output file 2 liquid>" << std::endl;
        return 1;
    }
    return 0;
}
`

output:
fftSize : 1048576
Total Time fftw: 41066 microseconds
create plan fftw: 1956 microseconds
execute fftw: 39098 microseconds
destroy fftw: 10 microseconds
Total Time Liquid: 143412 microseconds
create plan Liquid: 48338 microseconds
execute Liquid: 95056 microseconds
destroy Liquid: 18 microseconds

jgaeddert, 2022-10-15:
Yes, the reason why liquid-dsp's FFT is slower is because of several additional memory copy operations. These will get removed at some point. Grepping the source code indicates the only area that I'm using liquid's internal FFT is the [fftfilt](https://liquidsdr.org/api/fftfilt_crcf/), qdetector, and qpilotsync used for framing synchronization.

I should be able to pull these forward.

jgaeddert, 2022-10-21:
Should be up to date now.

jrmims, 2022-11-01:
Ok, just to confirm fftfilt, qdetector, and qpilotsync now use FFTW if
available and I should use FFTW directly if I need faster FFTs?

jgaeddert, 2022-11-01:
Yes, that is correct. They all use templated FFT methods and default to FFTW if installed. They fall back to the internal FFT methods if FFTW is not installed or if `configure` is run with the `--enable-fftoverride` option.

closed 2022-11-03 by jrmims