jgaeddert/liquid-dsp #296: Cannot use liquid-dsp functions, undefined reference error

kayakaan02, 2022-08-24:
### Discussed in https://github.com/jgaeddert/liquid-dsp/discussions/295

<div type='discussions-op-text'>

<sup>Originally posted by **kayakaan02** August 24, 2022</sup>
So when I try to use liquid-dsp with including liquid.h, any function that is associated with liquid-dsp is giving me undefined reference to [function]. As an example, I used the code on liquidsdr website, matrix operations, first example usage of printing a matrix(https://liquidsdr.org/doc/matrix/)
[image]
it gives me this error:  
```
C:\Users\thevo\AppData\Local\Temp\ccIGgj1j.o:asdf.c:(.text+0xfc): undefined reference to 'matrixf_print'
C:\Users\thevo\AppData\Local\Temp\ccIGgj1j.o:asdf.c:(.text+0x136): undefined reference to 'matrixf_ludecomp_doolittle'
collect2.exe: error: ld returned 1 exit status
```

Also I tried the commenting out in the troubleshooting part</div>

jgaeddert, 2022-08-24:
Sounds like a linier error. Are you linking with "-lliquid"? Please include your command line here.

kayakaan02, 2022-08-24:
> Sounds like a linker error. Are you linking with "-lliquid"? Please include your command line here.

Actually It seems like I haven't linked libliquid.a/libliquid.dll.a(created a new visual studio project and linked the libraries) so it was the problem. I am a little inexperienced so it was a newbie mistake.

closed 2022-08-24 by kayakaan02