jgaeddert/liquid-dsp #312: build: rename libliquid.a.1.5 to libliquid.a

eddsalkield, 2023-01-24:
Previous work https://github.com/jgaeddert/liquid-dsp/issues/133 https://github.com/jgaeddert/liquid-dsp/pull/180 set the soversion correctly, but resulted in `libliquid.a.1.5`, which should be called `libliquid.a` by convention. This issue seems to be blocking a merge into Alpine Linux: https://gitlab.alpinelinux.org/alpine/aports/-/merge_requests/40672

jgaeddert, 2023-03-18:
So the following change would suffice?

```diff
diff --git a/makefile.in b/makefile.in
index 3c0246d9b..8a243bb3f 100644
--- a/makefile.in
+++ b/makefile.in
@@ -1294,7 +1294,7 @@ install: all
        @echo ""
        mkdir -p $(DESTDIR)$(prefix)/include/liquid
        mkdir -p $(DESTDIR)$(libdir)
-       install -m 644 -p libliquid.${AR_LIB} $(DESTDIR)$(libdir)/libliquid.${AR_LIB}.${VERSION}
+       install -m 644 -p libliquid.${AR_LIB} $(DESTDIR)$(libdir)/libliquid.${AR_LIB}
        install -m 644 -p libliquid.${SH_LIB} $(DESTDIR)$(libdir)/libliquid.${SH_LIB}.${VERSION}
        install -m 644 -p include/liquid.h $(DESTDIR)$(prefix)/include/liquid/liquid.h
        ln -sf libliquid.${SH_LIB}.${VERSION} $(DESTDIR)$(libdir)/libliquid.${SH_LIB}
@@ -1321,7 +1321,7 @@ install: all
 uninstall:
        @echo "uninstalling..."
        $(RM) $(DESTDIR)$(prefix)/include/liquid/liquid.h
-       $(RM) $(DESTDIR)$(libdir)/libliquid.${AR_LIB}.${VERSION}
+       $(RM) $(DESTDIR)$(libdir)/libliquid.${AR_LIB}
        $(RM) $(DESTDIR)$(libdir)/libliquid.${SH_LIB}.${VERSION}
        $(RM) $(DESTDIR)$(libdir)/libliquid.${SH_LIB}.1
        $(RM) $(DESTDIR)$(libdir)/libliquid.${SH_LIB}
```

jgaeddert, 2023-03-18:
Testing with 88cd00d27a1c657275c646ceabfeeeafcf902925

jgaeddert, 2023-03-18:
I think this is resolved.
https://github.com/jgaeddert/liquid-dsp/actions/runs/4455847102/jobs/7825908768#step:8:9

closed 2023-03-18 by jgaeddert