jgaeddert/liquid-dsp #319: Compiler warnings in Raspberry-Pi3B

PavanMadushanka, 2023-03-23:
I see the following compiler warnings and notes while compiling the latest code in Raspberry-Pi (running on Raspbian Bullseye hf)

```sh
src/fft/src/asgram.proto.c:259:21: warning: writing 1 byte into a region of size 0 [-Wstringop-overflow=]
  259 |     ascii[_q->nfft] = '\0'; // append null character to end of string
      |     ~~~~~~~~~~~~~~~~^~~~~~
src/fft/src/asgram.proto.c:258:10: note: at offset [-9223372036854775808, 9223372036854775807] to an object with size at most 4294967295 declared here
  258 |     char ascii[_q->nfft+1];
      |          ^~~~~
gcc -g -O2    -Wall -fPIC -Wno-deprecated -Wno-deprecated-declarations -I. -Iinclude  -c -o src/fft/src/spgramf.o src/fft/src/spgramf.c
In file included from src/fft/src/spgramf.c:50:
src/fft/src/asgram.proto.c: In function ‘asgramf_print’:
src/fft/src/asgram.proto.c:259:21: warning: writing 1 byte into a region of size 0 [-Wstringop-overflow=]
  259 |     ascii[_q->nfft] = '\0'; // append null character to end of string
      |     ~~~~~~~~~~~~~~~~^~~~~~
src/fft/src/asgram.proto.c:258:10: note: at offset [-9223372036854775808, 9223372036854775807] to an object with size at most 4294967295 declared here
  258 |     char ascii[_q->nfft+1];
      |          ^~~~~
```

jgaeddert, 2023-03-25:
I'd like to reproduce this. What version of GCC are you running?

jgaeddert, 2023-03-25:
Ok, fresh install of Raspbian Bullseye (64-bit) running `gcc (Debian 10.2.1-6) 10.2.1 20210110` indeed reproduces this warning.

jgaeddert, 2023-03-25:
I think it's a gcc 10+ thing. Should be resolved in 445c352d163008cba82d1aa8aefb55d881929360...8f092ddcb9f12863c72c6579da7c99908946b3da

closed 2023-03-25 by jgaeddert

PavanMadushanka, 2023-03-27:
> I think it's a gcc 10+ thing. Should be resolved in [445c352...8f092dd](https://github.com/jgaeddert/liquid-dsp/compare/445c352d163008cba82d1aa8aefb55d881929360...8f092ddcb9f12863c72c6579da7c99908946b3da)

Thanks for the quick fix.