jgaeddert/liquid-dsp #322: license when using fftw

jhgit, 2023-05-01:
Should there be a mention in the readme or other documentation (configure --help) about using FFTW "infecting" this library with GPLv2?

jgaeddert, 2023-05-07:
> Should there be a mention in the readme or other documentation (configure --help) about using FFTW "infecting" this library with GPLv2?

My understanding is that _linking_ to libraries licensed under GPLv2 from source code under a different license is a bit of a gray area. FFTW isn't a dependency as liquid will run fine without it (albeit slower in several cases). I'm not distributing binaries, either. I am not a lawyer, though, so I can't comment definitively.

jhgit, 2023-05-15:
Using FFTW can make liquid-dsp a derivative work (regardless of how FFTW is incorporated into liquid-dsp - dynamically linked, statically linked or directly including the source code of FFTW into liquid-dsp and compiling together).  I don't know how much of a gray area it would be, but being gray should be enough to be considered by users who might be sensitive to the terms of a different license.  It's up to a legal authority to interpret the license terms and decide if the combination of FFTW and liquid-dsp is significant enough.

I think it would be prudent to at least mention the issue in the documentation that describes how FFTW can be used with liquid-dsp.

EighthOctave, 2023-06-07:
I've had concerns about this as well.  My understanding is LGPL allows for dynamic linking without being considered derivative work, but I think GPLv2 is pretty clear about any linking being considered derivative work which must also carry GPLv2 license.

Not a lawyer.

jgaeddert, 2023-06-19:
I'm not sure if this fits the bill, but I added some [notes in the README](https://github.com/jgaeddert/liquid-dsp#license) to that effect. Ideally, liquid wouldn't even need to link to these, but we're not quite there, yet.

I definitely appreciate the comments/feedback!

jimorc, 2023-11-21:
Actually, gnu.org is very specific about this. From [Frequently Asked Questions about the GNU Licenses](https://www.gnu.org/licenses/gpl-faq.en.html):

**If a library is released under the GPL (not the LGPL), does that mean that any software which uses it has to be under the GPL or a GPL-compatible license? ([#IfLibraryIsGPL](https://www.gnu.org/licenses/gpl-faq.en.html#IfLibraryIsGPL))**

Yes, because the program actually links to the library. As such, the terms of the GPL apply to the entire combination. The software modules that link with the library may be under various GPL compatible licenses, but the work as a whole must be licensed under the GPL. See also: [What does it mean to say a license is “compatible with the GPL”?](https://www.gnu.org/licenses/gpl-faq.en.html#WhatDoesCompatMean)

jhgit, 2023-11-27:
> I'm not sure if this fits the bill, but I added some [notes in the README](https://github.com/jgaeddert/liquid-dsp#license) to that effect. Ideally, liquid wouldn't even need to link to these, but we're not quite there, yet.
> 
> I definitely appreciate the comments/feedback!

Seems okay - adding your interpretation of one of the important caveats of linking with a GPL library (distribution of source code) is fine.  It'd be good to explicitly say that linking with fftw subjects liquid-dsp to all distribution terms of the GPLv2 license (or later).

Spelling nitpick: your use of 'vain' in the README should be spelled 'vein'.

But your additions are an improvement - it makes it somewhat clear that depending on third party libraries linked with liquid-dsp, the licensing terms can change from purely the MIT license terms.