jgaeddert/liquid-dsp #328: Unexpected behavior in the msequence module

sbhutch, 2023-07-21:
The documentation for the msequence module states that the default polynomial for a 5-bit sequence is x^5 + x^3 + 1, and the source code uses a lookup to derive g = 0x14 which looks correct. However, if I use the following code to create a msequence object and print the sequence:

```cpp
int main()
{
    msequence q = msequence_create_default(5);
    msequence_print(q);

    for (unsigned int i = 0; i < msequence_get_length(q); i++) {
        std::cout << msequence_advance(q) << " ";
    }
    std::cout << "\n";

    msequence_destroy(q);
}
```

I get the following output:

```
msequence: m=5 (n=31):
    shift register: 10000
    generator poly: 10010
1 0 1 0 1 1 1 0 1 1 0 0 0 1 1 1 1 1 0 0 1 1 0 1 0 0 1 0 0 0 0
```

The msequence state indicates a polynomial of x^5 + x^2 + 1 rather than x^5 + x^3 + 1. The sequence produced also backs this up when compared with the output from other libraries:

```py
from scipy import signal
import numpy as np

sequence, _ = signal.max_len_seq(
    nbits=5,
    state=[0, 0, 0, 0, 1],
    taps=[5, 2, 0],
)

print(np.flip(sequence))

# output: [1 0 1 0 1 1 1 0 1 1 0 0 0 1 1 1 1 1 0 0 1 1 0 1 0 0 1 0 0 0 0]
```

I also noticed when comparing to other libraries (scipy, commpy and gnuradio) that the sequence produced by liquid is in reverse order (hence np.flip). I am not sure if that actually matters or not though.

seonlee2013, 2023-07-21:
好啊，邮件收到啦！
Hi, your email is arrived safely

sbhutch, 2023-07-21:
Alright, I was using liquid 1.5.0 and these issues have been fixed since, that is amazing!

closed 2023-07-21 by sbhutch