jgaeddert/liquid-dsp #331: Processor and OS support AVX512, but GCC doesn't

thezboe, 2023-08-30:
It appears that the configure script checks for AVX512 support. In my case, the OS and processor support, but I have an ancient version of GCC that doesn't yet have the AVX512 options. The configure script correctly determines this: 
```
checking whether AVX512-F is supported by the processor... yes
checking whether AVX512-F is supported by the processor and OS... yes
checking whether C compiler accepts -mavx512f... no
configure: WARNING: Your processor and OS supports AVX512-F instructions but not your compiler, can you try another compiler?
checking whether AVX512-CD is supported by the processor... yes
checking whether AVX512-CD is supported by the processor and OS... yes
checking whether C compiler accepts -mavx512cd... no
configure: WARNING: Your processor and OS supports AVX512-CD instructions but not your compiler, can you try another compiler?
checking whether AVX512-PF is supported by the processor... no
checking whether AVX512-ER is supported by the processor... no
checking whether AVX512-VL is supported by the processor... yes
checking whether AVX512-VL is supported by the processor and OS... yes
checking whether C compiler accepts -mavx512vl... no
configure: WARNING: Your processor and OS supports AVX512-VL instructions but not your compiler, can you try another compiler?
checking whether AVX512-BW is supported by the processor... yes
checking whether AVX512-BW is supported by the processor and OS... yes
checking whether C compiler accepts -mavx512bw... no
configure: WARNING: Your processor and OS supports AVX512-BW instructions but not your compiler, can you try another compiler?
checking whether AVX512-DQ is supported by the processor... yes
checking whether AVX512-DQ is supported by the processor and OS... yes
checking whether C compiler accepts -mavx512dq... no
configure: WARNING: Your processor and OS supports AVX512-DQ instructions but not your compiler, can you try another compiler?
checking whether AVX512-IFMA is supported by the processor... no
checking whether AVX512-VBMI is supported by the processor... no
```

But, the resulting makefile still tries to use the flags, resulting in error.
```
gcc -g -O2  -mavx512f  -Wall -fPIC -Wno-deprecated -Wno-deprecated-declarations -I. -Iinclude  -c -o src/libliquid.o src/libliquid.c
gcc: error: unrecognized command line option \u2018-mavx512f\u2019
make: *** [src/libliquid.o] Error 1
```

thezboe, 2023-08-30:
It appears that configure.ac should be checking `ax_cv_support_avx512f_ext` instead of `ax_cv_have_avx512f_ext`?

jgaeddert, 2023-09-01:
Good catch. As much as I hate to admit it, this is where CMake is useful...

pfeatherstone, 2023-09-01:
Have you looked at zigbuild ? Or zig for that matter. It has a nice embedded build system and can cross compile (C, C++ and Zig) to tons of targets using the same binary.

JayKickliter, 2023-09-01:
> Have you looked at zigbuild ? Or zig for that matter. It has a nice embedded build system and can cross compile (C, C++ and Zig) to tons of targets using the same binary.

Zig is pretty awesome for cross-compiling C, and as much as I hate CMake/autotools, using anything else is huge pain for downstream users.

jgaeddert, 2023-09-01:
I *have* looked into zig! Supports built-in testing as well!

lukeprince20, 2023-09-08:
> Good catch. As much as I hate to admit it, this is where CMake is useful...

CMake is great (IMO)! But checking for microarchitectures is still pretty niche. Earlier this year I added https://github.com/jgaeddert/liquid-dsp/pull/314 which added CMake build support for the previous release version. In it I used Google's awesome [cpu_features](https://github.com/google/cpu_features) library for detection of supported CPU instruction sets. You can see how I hooked it into the build configuration [here](https://github.com/lukeprince20/liquid-dsp/blob/cmake_build/CMakeLists.txt#L252-L329). If there's interest I can resurrect this PR for the current release version 1.6.