jgaeddert/liquid-dsp #348: Compiling  liquid-dsp for  Zynq Processor

iottrends, 2023-11-17:
Hi 
I have built a sample ofdm application using liquid dsp it works on x86 platform. I am trying to benchmark various ofdm configurations .
./test2_ofdm 512 32

 OFDM = 512 , CP =32 
 
 total time avg:294us, AVG(odfmframegen)= 184, AVG(OFDM_write)= 110

 Min(ofdmframegen) = 137 us Max(ofdmframegen)= 580 us

 Min(OFDM_write) = 85 us Max(OFDM_write)= 85 us 
 
 total time avg:318us, AVG(odfmframegen)= 200, AVG(OFDM_write)= 118

 Min(ofdmframegen) = 164 us Max(ofdmframegen)= 593 us

 Min(OFDM_write) = 97 us Max(OFDM_write)= 97 us 
 
 now i want to take this C code to plutosdr and hence trying to cross compile it.

took a fresh git clone.
1.  ./configure --build x86_64-pc-linux-gnu --host=arm-linux-gnueabi
2. output:
abh@abh-VirtualBox:~/Documents/dsp_code/Zynq_executable/liquid-dsp$ ./configure --build x86_64-pc-linux-gnu --host=arm-linux-gnueabi
checking for arm-linux-gnueabi-gcc... arm-linux-gnueabi-gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... yes
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether arm-linux-gnueabi-gcc accepts -g... yes
checking for arm-linux-gnueabi-gcc option to accept ISO C89... none needed
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for a BSD-compatible install... /usr/bin/install -c
checking for arm-linux-gnueabi-ranlib... arm-linux-gnueabi-ranlib
checking for error_at_line... yes
checking how to run the C preprocessor... arm-linux-gnueabi-gcc -E
checking for egrep... /usr/bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for stdlib.h... (cached) yes
checking for GNU libc compatible malloc... no
checking for stdlib.h... (cached) yes
checking for GNU libc compatible realloc... no
checking for main in -lc... yes
checking for main in -lm... yes
checking for malloc... yes
checking for realloc... yes
checking for free... yes
checking for memset... yes
checking for memmove... yes
checking for sinf... yes
checking for cosf... yes
checking for expf... yes
checking for cargf... yes
checking for cexpf... yes
checking for crealf... yes
checking for cimagf... yes
checking for sqrtf... yes
checking stdio.h usability... yes
[... 375 lines cut ...]
7. abh@abh-VirtualBox:~/Documents/dsp_code/Zynq_executable/liquid-dsp$ du -sh *
4.0K	aclocal.m4
708K	autom4te.cache
72K	autotest
60K	bench
4.0K	bootstrap.sh
12K	config.h
8.0K	config.h.in
96K	config.log
32K	config.status
212K	configure
12K	configure.ac
1.1M	examples
20K	HISTORY
640K	include
5.2M	libliquid.a
4.0K	library.json
4.0K	LICENSE
64K	makefile
64K	makefile.in
12K	README.md
904K	sandbox
220K	scripts
12M	src
4.0K	TROUBLESHOOTING

jgaeddert, 2023-11-24:
I am not an expert on cross-compiling for ARM, but it seems as though there's a conflict with the vector floating-point coprocessor, maybe? Did you re-build from scratch? Your `make` line is showing linking to a shared object; can you show what the output of the build process is for each object file?

piaooo, 2024-04-11:
参考https://github.com/jgaeddert/liquid-dsp/issues/146
I can build on pluto-0.38.sysroot:
```
git clone https://github.com/jgaeddert/liquid-dsp.git
cd liquid-dsp
```
edit configure.ac
ARCH_OPTION="-ffast-math -mcpu=cortex-a7 -mfloat-abi=hard -mfpu=neon-vfpv4";;
to: 
ARCH_OPTION="-ffast-math -march=armv7-a -mfpu=neon -mfloat-abi=hard";;
```
./bootstrap.sh
./configure --prefix=/root/pluto-0.38.sysroot/usr/ --host=arm-linux-gnueabihf
make
make install
```