jgaeddert/liquid-dsp #350: Question: Is there a function like the scipy.signal.resample_poly that remove the phase shift after decimation?

yonghuier, 2023-12-16:
I needed to use an externally designed low-pass filter for downsampling, so I used the `firdecim_crcf_create` function, but I found that it was actually similar to the `upfirdn` function, with a phase shift after downsampling. 

But what I want is downsampling without phase shift, like the `scipy.signal.resample_poly` function does. 

Is there a function that can be called directly, or what can I do to avoid phase shift?

seonlee2013, 2023-12-16:
好啊，邮件收到啦！
Hi, your email is arrived safely

jgaeddert, 2024-01-27:
The delay of the decimator will depend upon the filter coefficients themselves. Normally, an FIR filter has symmetric coefficients and has a length $2Mm+1$ in which case the output delay is $Mm$ samples, where $M$ is the decimation rate and $m$ is the filter semi-length. But if the filter coefficients are _not_ symmetric or if the filter length follows a different length, the delay will be something different. Unfortunately the decimator cannot compensate for this automatically since it doesn't know the filter response. The delay could even be a fraction of a sample.

If the filter does have a length $2Mm+1$ then the output delay will be $m$ samples; this is one of the reasons designing a filter of this length is convenient. If this is the case, you can then just ignore the first $m$ output samples (e.g. after decimation), and then flush $m$ blocks of $M$ samples each through the filter to compensate for that delay.