jgaeddert/liquid-dsp #354: Rician fading - how to measure K and omega

pfeatherstone, 2024-02-13:
In https://github.com/jgaeddert/liquid-dsp/blob/master/examples/ricek_channel_example.c you generate rician flat fading using the linear filter technique. This is cool.
How do I measure `K` and `omega` from the generated fading gains stored in array `y` ?

I tried using:
```
    // Measure K and omega and check
    float s12 = 0;
    float s22 = 0;

    for (i = 0 ; i < num_samples ; ++i)
    {
        float tmp = cabsf(y[i]);
        s12 += tmp;
        s22 += tmp*tmp;
    }

    float mu        = s12 / num_samples;
    float std2      = s22 / num_samples - mu*mu;
    float K_m       = mu*mu/(2*std2);
    float omega_m   = mu*mu+2*std2;
    printf("Actual K %f measured %f - omega %f measured %f\n", K, K_m, omega, omega_m);
```

But they don't match. 
@jgaeddert Can you shed some light?
Thank you

pfeatherstone, 2024-02-13:
Looking at the Wikipedia page on parameter estimation of rice distribution, it's not that simple

jgaeddert, 2024-02-20:
It's funny; my MS thesis was on this exact topic. Well, actually Rice, Namakami-m, and Weibull fading parameter estimation. While it's true that Rice-K parameter estimation is the most challenging (and there's no closed-form solution), there are some simple tricks to play that can help.

Let me dust off some old work I did and see what suggestions I might come up with.

What values/ranges of K and Omega are you concerned with?

pfeatherstone, 2024-02-20:
What i've been doing is compute a histogram, which measures a discrete probability distribution. Then I use https://docs.scipy.org/doc/scipy/reference/generated/scipy.optimize.minimize.html#scipy.optimize.minimize to find parameters K and omega where `fun` is the difference between the Rice PDF and the histogram evaluated at the histogram bins.