jgaeddert/liquid-dsp #359: OFDM multipath fading problem

piaooo, 2024-04-11:
Hi
I try built a sample ofdm application using liquid-dsp work on plutosdr.I want to deal with multipath problem and I saw an article [Using the cyclic prefix to solve the OFDM multipath fading problem](https://forum.huawei.com/enterprise/en/using-the-cyclic-prefix-to-solve-the-ofdm-multipath-fading-problem/thread/667242838481584128-667213872962088960). I refer to  [ofdmflexframesync_example.c](https://github.com/jgaeddert/liquid-dsp/blob/master/examples/ofdmflexframesync_example.c), when i add `channel_cccf_add_multipath(channel, NULL, hc_len)` and set hc_len = 8, whatever i change cyclic prefix length value it cant get payload with no error.

Does using Equalization help with multipath? I saw that Symbol Tracker can be recovered. Where should I add it to the above code?

jgaeddert, 2024-04-12:
Can you provide a full example with the changes you made to include the multi-path component? The `ofdmflexframesync` includes equalization; however you might need to include more pilot subcarriers depending upon the severity of the multipath. Increasing the cyclic prefix length will prevent inter-symbol interference, but will not directly compensate for the channel effects.

piaooo, 2024-04-13:
Thanks for reply

I have tried modifying the number of subcarriers but the bit errors will not be eliminated.

I just added channel_cccf_add_multipath() to this example and also provided the hc_len modifiable parameter. Exactly when hc_len is set to 5, payloads are invalid. you can try 
```bash
gcc test.c -lliquid -o multipath_test
./multipath_test -l 5
```
```C
#include <stdio.h>
#include <stdlib.h>
#include <math.h>
#include <string.h>
#include <getopt.h>
#include <time.h>

#include "liquid/liquid.h"

void usage()
{
    printf("ofdmflexframesync_example [options]\n");
    printf(" -h        : print usage\n");
    printf(" -s  <snr> : signal-to-noise ratio [dB], default: 20\n");
    printf(" -F <freq> : carrier frequency offset, default: 0.002\n");
    printf(" -M  <num> : number of subcarriers (must be even), default: 64\n");
    printf(" -C  <len> : cyclic prefix length, default: 16\n");
    printf(" -n  <len> : payload length [bytes], default: 120\n");
    printf(" -m  <mod> : modulation scheme (qpsk default)\n");
    liquid_print_modulation_schemes();
    printf(" -v  <crc> : data integrity check: crc32 default\n");
    liquid_print_crc_schemes();
    printf(" -c  <fec> : coding scheme (inner): h74 default\n");
    printf(" -k  <fec> : coding scheme (outer): none default\n");
    liquid_print_fec_schemes();
    printf(" -d       : enable debugging\n");
}

// callback function
int callback(unsigned char *  _header,
             int              _header_valid,
             unsigned char *  _payload,
             unsigned int     _payload_len,
             int              _payload_valid,
             framesyncstats_s _stats,
             void *           _userdata);

int main(int argc, char*argv[])
{
    //srand(time(NULL));

    // options
    unsigned int      M           = 64;                 // number of subcarriers
    unsigned int      cp_len      = 16;                 // cyclic prefix length
    unsigned int      taper_len   = 4;                  // taper length
    unsigned int      payload_len = 120;                // length of payload (bytes)
    modulation_scheme ms          = LIQUID_MODEM_QAM16;  // modulation scheme
    fec_scheme        fec0        = LIQUID_FEC_NONE;    // inner code
    fec_scheme        fec1        = LIQUID_FEC_SECDED7264; // outer code
    crc_scheme        check       = LIQUID_CRC_32;      // validity check
    float             noise_floor = -80.0f;             // noise floor [dB]
    float             SNRdB       = 20.0f;              // signal-to-noise ratio [dB]
    float             dphi        = 0.02f;              // carrier frequency offset
    int               debug       =  0;                 // enable debugging?
    int               hc_len      = 4;
    // get options
    int dopt;
    while((dopt = getopt(argc,argv,"uhds:F:M:C:n:m:v:c:k:l:")) != EOF){
        switch (dopt) {
        case 'u':
[... 103 lines cut ...]
    unsigned int i;

    // print header data to standard output
    printf("  header rx  :");
    for (i=0; i<8; i++)
        printf(" %.2X", _header[i]);
    printf("\n");

    // print payload data to standard output
    printf("  payload rx :");
    for (i=0; i<_payload_len; i++) {
        printf(" %.2X", _payload[i]);
        if ( ((i+1)%26)==0 && i !=_payload_len-1 )
            printf("\n              ");
    }
    printf("\n");

    // count errors in received payload and print to standard output
    unsigned char * payload_tx = (unsigned char*) _userdata;
    unsigned int num_errors = count_bit_errors_array(_payload, payload_tx, _payload_len);
    printf("  bit errors : %u / %u\n", num_errors, 8*_payload_len);

    return 0;
}
```

piaooo, 2024-04-13:
> Can you provide a full example with the changes you made to include the multi-path component? The `ofdmflexframesync` includes equalization; however you might need to include more pilot subcarriers [...]

Thanks for reply

I have tried modifying the number of subcarriers but the bit errors will not be eliminated.

I just added channel_cccf_add_multipath() to this example and also provided the hc_len modifiable parameter. Exactly when hc_len is set to 5, payloads are invalid. you can try
```bash
gcc test.c -lliquid -o multipath_test
./multipath_test -l 5
```
```c
#include <stdio.h>
#include <stdlib.h>
#include <math.h>
#include <string.h>
#include <getopt.h>
#include <time.h>

#include "liquid/liquid.h"

void usage()
{
    printf("ofdmflexframesync_example [options]\n");
    printf(" -h        : print usage\n");
    printf(" -s  <snr> : signal-to-noise ratio [dB], default: 20\n");
    printf(" -F <freq> : carrier frequency offset, default: 0.002\n");
    printf(" -M  <num> : number of subcarriers (must be even), default: 64\n");
    printf(" -C  <len> : cyclic prefix length, default: 16\n");
    printf(" -n  <len> : payload length [bytes], default: 120\n");
    printf(" -m  <mod> : modulation scheme (qpsk default)\n");
    liquid_print_modulation_schemes();
    printf(" -v  <crc> : data integrity check: crc32 default\n");
    liquid_print_crc_schemes();
    printf(" -c  <fec> : coding scheme (inner): h74 default\n");
    printf(" -k  <fec> : coding scheme (outer): none default\n");
    liquid_print_fec_schemes();
    printf(" -d       : enable debugging\n");
}

// callback function
int callback(unsigned char *  _header,
             int              _header_valid,
             unsigned char *  _payload,
             unsigned int     _payload_len,
             int              _payload_valid,
             framesyncstats_s _stats,
             void *           _userdata);

int main(int argc, char*argv[])
{
    //srand(time(NULL));

    // options
    unsigned int      M           = 64;                 // number of subcarriers
    unsigned int      cp_len      = 16;                 // cyclic prefix length
    unsigned int      taper_len   = 4;                  // taper length
    unsigned int      payload_len = 120;                // length of payload (bytes)
    modulation_scheme ms          = LIQUID_MODEM_QAM16;  // modulation scheme
    fec_scheme        fec0        = LIQUID_FEC_NONE;    // inner code
    fec_scheme        fec1        = LIQUID_FEC_SECDED7264; // outer code
    crc_scheme        check       = LIQUID_CRC_32;      // validity check
    float             noise_floor = -80.0f;             // noise floor [dB]
    float             SNRdB       = 20.0f;              // signal-to-noise ratio [dB]
    float             dphi        = 0.02f;              // carrier frequency offset
    int               debug       =  0;                 // enable debugging?
    int               hc_len      = 4;
    // get options
    int dopt;
    while((dopt = getopt(argc,argv,"uhds:F:M:C:n:m:v:c:k:l:")) != EOF){
[... 105 lines cut ...]
    unsigned int i;

    // print header data to standard output
    printf("  header rx  :");
    for (i=0; i<8; i++)
        printf(" %.2X", _header[i]);
    printf("\n");

    // print payload data to standard output
    printf("  payload rx :");
    for (i=0; i<_payload_len; i++) {
        printf(" %.2X", _payload[i]);
        if ( ((i+1)%26)==0 && i !=_payload_len-1 )
            printf("\n              ");
    }
    printf("\n");

    // count errors in received payload and print to standard output
    unsigned char * payload_tx = (unsigned char*) _userdata;
    unsigned int num_errors = count_bit_errors_array(_payload, payload_tx, _payload_len);
    printf("  bit errors : %u / %u\n", num_errors, 8*_payload_len);

    return 0;
}
```

piaooo, 2024-06-06:
Hello,

Do you have any suggestions? I have added multipath interference in the channel simulator, but the results are still not satisfactory.

Thank you!

closed 2024-07-01 by piaooo