jgaeddert/liquid-dsp #363: make check fails on Segmentation fault in mingw Windows 11

talhafaizurrahman, 2024-05-27:
I have managed to install liquid dsp but when i do "make check" it stops on segmentation fault. 

msresamp_crcf_num_output_6:
testing msresamp_crcf_get_num_output() with r=2980.96
max_input : 64, max_output : 763141
make: *** [makefile:1337: check] Segmentation fault

I am on mingw64 in windows. Any idea what could be wrong here?

make install works fine. 
make example works fine. 

Moreover, I am also trying to link library in code blocks IDE but failing everytime. I can share the results if you like. 
Thanks

jgaeddert, 2024-06-08:
In order for me to help with this, I'll need a little more information. What specific version of the code are you using (e.g. git hash)? Does it fail every time you run "make check" (a different random seed is provided each time)? I'm not an expert on running in windows; what compiler/version are you using?

talhafaizurrahman, 2024-06-18:
Hi Gaeddert, 
Apologies for the delayed response. Installing liquid on windows was not a good idea. What i did is, i installed it on WSL and use VScode to access the libraries that are installed on WSL. So many tutorials are available on this. Thank you for making such a cool DSP library.

closed 2024-06-18 by talhafaizurrahman