jgaeddert/liquid-dsp #376: [Question] FSK synchronization

pfeatherstone, 2024-07-25:
I have a 4-FSK signal that i want to demodulate. It doesn't have a preamble or anything like that to assist with synchronization. The `fskdem` object requires perfect symbol timing. How can we realise symbol timing?

xerpi, 2024-07-25:
This might help: https://liquidsdr.org/doc/symtrack/

pfeatherstone, 2024-07-28:
Does that work with FSK? Not sure

pfeatherstone, 2024-07-31:
Yeah symtrack is only designed to work with [modem](https://liquidsdr.org/doc/modem/) objects which doesn't include FSK

xerpi, 2024-08-01:
I think you can still use `symsync` (and `symtrack`?). Just make sure to pass the samples through the Gaussian filter (`LIQUID_FIRFILT_GMSKRX`) after FM demodulation. Check this example: https://github.com/jgaeddert/liquid-dsp/blob/master/examples/gmsk_tracking_example.c#L62

Here's more info on what makes GMSK different regarding the matched filter: https://dsp.stackexchange.com/a/74319
> GMSK doesn't just use a Gaussian filter instead of a raised-cosine filter. It uses a Gaussian filter on the phase, **_before_** applying it to the modulator. This makes it a nonlinear operation.
> When a raised-cosine filter is applied to some modulated signal, it is applied **_after_** modulation, as a linear operation.

pfeatherstone, 2024-08-02:
I've tried using symsync for FSK4 and it's not working well at all.

pfeatherstone, 2024-08-02:
I'm using something similar to `freqdem` or GNURadio's Quadrature demod block to basically convert FSK4 to ASK4 then I need some kind of symbol synchronization and PLL. In this case the error in the amplitude equals to frequency error, which then needs to be fed to a PLL loop. So i don't think i can simply re-use an ASK4 demod algorithm.