jgaeddert/liquid-dsp #378: Very large number of samples produced

Mas313, 2024-08-25:
Hi,
When running example of gmsk frame sync with k (number of samples per symbol)   = 203 with remaining parameter like filter length m=2 and BT = 0.5 the number of samples produced are quite large arround 1779261.
`unsigned int num_samples = gmskframegen_getframelen(fg) + 800;` .
This number of samples are  producing large amount of delay when example is run standalone and not able to decode when did practically sending through plutosdr with 2 Msps. Any idea what could be wrong.
It works with k=2 with 2 Msps which is not practical for real scenario.
Can't liquiddsp handle large k values.

xerpi, 2024-08-26:
Isn't the number of samples per symbol (sps) too high (203)?
From what I've seen, the usual is in the order of 5-20 sps.

Mas313, 2024-08-27:
I want to demodulate AIS waveform upconverted to plutosdr frequency range (483.07Mhz). The data rate is 9600 bps. So when sampling at 2 Msps the samples per symbol are 208, which is not working with liquid dsp. Ingact anything above 10 for sps not working. I don't understand how gnuradio decod3s AIS then.

xerpi, 2024-08-27:
I think you can first convert to baseband ([mixing](https://liquidsdr.org/doc/nco/)) and then apply decimation (+FIR to get rid of aliasing) to around 5-20 sps.
Using a Polyphase decimator is a very efficient way to do decimation+FIR since it will only apply the FIR filter after decimation. In liquid-dsp I think this object implements that: https://liquidsdr.org/doc/firdecim/

Mas313, 2024-08-27:
Thanks, can you tell for tx also I need to use 208 samples per symbol from  2msps/9600, which liquiddsp not handle correctly but gnuradio in an open source project of AIS generator uses 208 samples per symbol and works correctly. So what I should do for tx part. Thanks for your time.

xerpi, 2024-08-27:
For TX you can probably interpolate (aka upsample) to the target SPS and then mix from baseband to passband.
Check this liquid-dsp object: https://liquidsdr.org/doc/firinterp/

Mas313, 2024-08-27:
Thanks for quick response.
If say i initially used value of 10 sps and generate samples for gmsk, the resulting data rate would be 200 000 bps with sample rate of 2 Msps. Now to achieve 9600 bps how to proceed.

xerpi, 2024-08-27:
> Thanks for quick response. If say i initially used value of 10 sps and generate samples for gmsk, the resulting data rate would be 200 000 bps with sample rate of 2 Msps. Now to achieve 9600 bps how to proceed.

The samples per symbol would increase (like from 20 to 200) but the baud/data rate would be the same.
In other words, each symbol would have more samples, but the symbol rate would be the same.

Mas313, 2024-08-27:
In your mentioned solution, the number of samples per symbol in case of tx and rx which are input to gmsk modulator and demodulator would be  low( 4 to 10 ) which is supported by liquid dsp. However the samples per symbol over the air wiuld be high. So would it benefit. I thought somehow if liquiddsp handles high number if samples then more frames would be decode.