jgaeddert/liquid-dsp #386: liquid-dsp pypi build instructions

justingray, 2024-12-17:
Hello,

I was trying to build your python to C++ binding as described here https://pypi.org/project/liquid-dsp/ but I was unable to find any of the necessary pybind wrappers or other necessary files to do a `pip install .` Perhaps I'm just missing some nuance, my eventual goal is to flesh out some of the bindings for symtrack.

Thank you

seonlee2013, 2024-12-17:
好啊，邮件收到啦！
Hi, your email is arrived safely

justingray, 2024-12-17:
I realized I probably need to be on `bind-scikit-build-dev` still having issues building, but that could be a personal problem.

justingray, 2024-12-19:
the build instruction was `pip install .` I only had success running `python -m build` to generate my own whl file

closed 2024-12-19 by justingray