jgaeddert/liquid-dsp #397: Build Error On Raspberry Pi

Mas313, 2025-02-14:
On Raspberry Pi 4 running `./bootstrap.sh` and `./configure` produces no error while on running `make` i get error as mentioned below. Please tell how to properly install liquid dsp on Raspberry Pis:

```
gc/src/agc_rrrf.o src/agc/src/agc_rrrf.c
gcc -g -O2  -ffast-math -mcpu=cortex-a7 -mfloat-abi=hard -mfpu=neon-vfpv4  -Wall -fPIC -Wno-deprecated -Wno-deprecated-declarations -I. -I./include  -c -o src/audio/src/cvsd.o src/audio/src/cvsd.c
gcc: error: unrecognized command-line option ‘-mfloat-abi=hard’
gcc: error: unrecognized command-line option ‘-mfloat-abi=hard’
gcc: error: unrecognized command-line option ‘-mfloat-abi=hard’
gcc: error: unrecognized command-line option ‘-mfloat-abi=hard’
gcc: error: unrecognized command-line option ‘-mfpu=neon-vfpv4’
make: *** [<builtin>: src/audio/src/cvsd.o] Error 1
make: *** Waiting for unfinished jobs....
gcc: error: unrecognized command-line option ‘-mfpu=neon-vfpv4’
make: *** [<builtin>: src/libliquid.o] Error 1
gcc: error: unrecognized command-line option ‘-mfpu=neon-vfpv4’
make: *** [<builtin>: src/agc/src/agc_rrrf.o] Error 1
gcc: error: unrecognized command-line option ‘-mfpu=neon-vfpv4’
make: *** [<builtin>: src/agc/src/agc_crcf.o] Error 1
```

jgaeddert, 2025-02-17:
I can confirm this is the case. Looking into this now.

jgaeddert, 2025-03-15:
Not sure what change with the pi4, but the old compile flags seem to break it. I reverted to just disabling these in 3a5e1f578ad5e73d7665e71781e764765608c2a2 and it passes the tests on my end.