jgaeddert/liquid-dsp #403: Cross compiling from Linux to Windows using mingw-w64 is extremely difficult via cmake

pclov3r, 2025-04-09:
Per the title. Cross compiling from Linux to Windows using mingw-w64 via cmake is extremely difficult. There is several issues.

All the try runs attempt to run the cross compiled code natively and as a result this results in major problems or Segmentation Faulting.

`target_link_libraries(${LIBNAME} PRIVATE c m)` is a problem in the  CMakeLists File

This can be fixed with something like:

```
if(NOT WIN32)
    target_link_libraries(${LIBNAME} PRIVATE c m)
endif()

```

No easy way to specify what CPU Acceleration options without adding Cmake Cache Values to your toolchain file. 

Hopefully there is a way to clean this up :)

seonlee2013, 2025-04-09:
好啊，邮件收到啦！
Hi, your email is arrived safely

jgaeddert, 2025-04-15:
Thanks for the info! Can you provide a set of instructions I can use for testing this on my end? In the [cmake-dev branch](https://github.com/jgaeddert/liquid-dsp/tree/cmake-dev) there are a few changes that should help with cross-compiling, specifically specifying the SIMD instructions.

    cmake -D ENABLE_SIMD=ON -D FIND_SIMD=OFF -D C_AVX512_FOUND=ON -D C_AVX512_FLAGS="<custom flags>" ..

This tells cmake to enable SIMD, but don't search for a set of instructions on the local build system, specify that we are using AVX512 and provide a set of compiler flags.

pclov3r, 2025-04-21:
Sorry for the late reply missed the email.

To build for windows i did the

```
if(NOT WIN32)
    target_link_libraries(${LIBNAME} PRIVATE c m)
endif()

```

change in the main cmakelists 

The toolchain file looked like this

```
# Set the target system name
set(CMAKE_SYSTEM_NAME Windows)
set(CMAKE_SYSTEM_PROCESSOR x86_64) # Architecture

# Specify the cross-compilers
SET(CMAKE_C_COMPILER x86_64-w64-mingw32-gcc)
SET(CMAKE_CXX_COMPILER x86_64-w64-mingw32-g++)
SET(CMAKE_RC_COMPILER x86_64-w64-mingw32-windres)

# Set the find root path modes for cross-compiling
set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_PACKAGE ONLY)

message(STATUS "Cross-compiling for Windows: Caching try_run results for ALL SIMD checks.")

# --- C Language SIMD Checks (Exit Code 0=Success, 1=Failure) ---
set(C_HAS_SSE4_1_EXITCODE   0 CACHE INTERNAL "try_run exit code for C SSE4.1")
set(C_HAS_SSE4_2_EXITCODE   0 CACHE INTERNAL "try_run exit code for C SSE4.2")
set(C_HAS_AVX_1_EXITCODE    0 CACHE INTERNAL "try_run exit code for C AVX_1")
set(C_HAS_AVX_2_EXITCODE    0 CACHE INTERNAL "try_run exit code for C AVX_2")
set(C_HAS_AVX_3_EXITCODE    0 CACHE INTERNAL "try_run exit code for C AVX_3")
set(C_HAS_AVX2_1_EXITCODE   1 CACHE INTERNAL "try_run exit code for C AVX2_1")
set(C_HAS_AVX2_2_EXITCODE   1 CACHE INTERNAL "try_run exit code for C AVX2_2")
set(C_HAS_AVX512_1_EXITCODE 1 CACHE INTERNAL "try_run exit code for C AVX512_1")
set(C_HAS_AVX512_2_EXITCODE 1 CACHE INTERNAL "try_run exit code for C AVX512_2")
set(C_HAS_AVX512_3_EXITCODE 1 CACHE INTERNAL "try_run exit code for C AVX512_3")
set(C_HAS_NEON_1_EXITCODE   1 CACHE INTERNAL "try_run exit code for C NEON_1")
set(C_HAS_NEON_2_EXITCODE   1 CACHE INTERNAL "try_run exit code for C NEON_2")
set(C_HAS_NEON_3_EXITCODE   1 CACHE INTERNAL "try_run exit code for C NEON_3")
set(C_HAS_NEON_4_EXITCODE   1 CACHE INTERNAL "try_run exit code for C NEON_4")
set(C_HAS_ALTIVEC_1_EXITCODE 1 CACHE INTERNAL "try_run exit code for C ALTIVEC_1")
set(C_HAS_ALTIVEC_2_EXITCODE 1 CACHE INTERNAL "try_run exit code for C ALTIVEC_2")
set(C_HAS_ALTIVEC_3_EXITCODE 1 CACHE INTERNAL "try_run exit code for C ALTIVEC_3")

# --- CXX Language SIMD Checks (Exit Code 0=Success, 1=Failure) ---
set(CXX_HAS_SSE4_1_EXITCODE   0 CACHE INTERNAL "try_run exit code for C SSE4.1")
set(CXX_HAS_SSE4_2_EXITCODE   0 CACHE INTERNAL "try_run exit code for C SSE4.2")
set(CXX_HAS_AVX_1_EXITCODE    0 CACHE INTERNAL "try_run exit code for C AVX_1")
set(CXX_HAS_AVX_2_EXITCODE    0 CACHE INTERNAL "try_run exit code for C AVX_2")
set(CXX_HAS_AVX_3_EXITCODE    0 CACHE INTERNAL "try_run exit code for C AVX_3")
set(CXX_HAS_AVX2_1_EXITCODE   1 CACHE INTERNAL "try_run exit code for C AVX2_1")
set(CXX_HAS_AVX2_2_EXITCODE   1 CACHE INTERNAL "try_run exit code for C AVX2_2")
set(CXX_HAS_AVX512_1_EXITCODE 1 CACHE INTERNAL "try_run exit code for C AVX512_1")
set(CXX_HAS_AVX512_2_EXITCODE 1 CACHE INTERNAL "try_run exit code for C AVX512_2")
set(CXX_HAS_AVX512_3_EXITCODE 1 CACHE INTERNAL "try_run exit code for C AVX512_3")
set(CXX_HAS_NEON_1_EXITCODE   1 CACHE INTERNAL "try_run exit code for C NEON_1")
set(CXX_HAS_NEON_2_EXITCODE   1 CACHE INTERNAL "try_run exit code for C NEON_2")
set(CXX_HAS_NEON_3_EXITCODE   1 CACHE INTERNAL "try_run exit code for C NEON_3")
set(CXX_HAS_NEON_4_EXITCODE   1 CACHE INTERNAL "try_run exit code for C NEON_4")
set(CXX_HAS_ALTIVEC_1_EXITCODE 1 CACHE INTERNAL "try_run exit code for C ALTIVEC_1")
set(CXX_HAS_ALTIVEC_2_EXITCODE 1 CACHE INTERNAL "try_run exit code for C ALTIVEC_2")
set(CXX_HAS_ALTIVEC_3_EXITCODE 1 CACHE INTERNAL "try_run exit code for C ALTIVEC_3")

# Also cache the corresponding TRYRUN_OUTPUT variables (usually empty for simple checks)
foreach(CHECK_VAR
    C_HAS_SSE4_1 C_HAS_AVX_1 C_HAS_AVX_2 C_HAS_AVX_3 C_HAS_AVX2_1 C_HAS_AVX2_2
    C_HAS_AVX512_1 C_HAS_AVX512_2 C_HAS_AVX512_3
    C_HAS_NEON_1 C_HAS_NEON_2 C_HAS_NEON_3 C_HAS_NEON_4
    C_HAS_ALTIVEC_1 C_HAS_ALTIVEC_2 C_HAS_ALTIVEC_3
    CXX_HAS_SSE4_1 CXX_HAS_AVX_1 CXX_HAS_AVX_2 CXX_HAS_AVX_3 CXX_HAS_AVX2_1 CXX_HAS_AVX2_2
    CXX_HAS_AVX512_1 CXX_HAS_AVX512_2 CXX_HAS_AVX512_3
    CXX_HAS_NEON_1 CXX_HAS_NEON_2 CXX_HAS_NEON_3 CXX_HAS_NEON_4
    CXX_HAS_ALTIVEC_1 CXX_HAS_ALTIVEC_2 CXX_HAS_ALTIVEC_3
)
    set(${CHECK_VAR}_EXITCODE__TRYRUN_OUTPUT "" CACHE INTERNAL "try_run output for ${CHECK_VAR}")
endforeach()
```

Then a cmake command like

cmake -DBUILD_EXAMPLES=OFF -DBUILD_AUTOTESTS=OFF -DBUILD_BENCHMARKS=OFF -DCMAKE_TOOLCHAIN_FILE=(windows-cross-compile-toolchain-liquid-dsp) ..

the examples, autotests, and benchmarks all will not build correctly so i just disabled all of them to try and get the core library built.

pclov3r, 2025-04-24:
Playing around here is some stuff to get a build going a bit easier. 

cmake/summary.cmake file

```
# cmake/summary.cmake

# status report function
# Automatically align right column and selects text based on condition.
function(liquid_status text)
  set(status_cond)
  set(status_then)
  set(status_else)

  set(status_current_name "cond")
  foreach(arg ${ARGN})
    if(arg STREQUAL "THEN")
      set(status_current_name "then")
    elseif(arg STREQUAL "ELSE")
      set(status_current_name "else")
    else()
      list(APPEND status_${status_current_name} ${arg})
    endif()
  endforeach()

  if(DEFINED status_cond)
    # Pad the 'text' part for alignment
    set(status_placeholder_length 23) # Length used for padding
    string(RANDOM LENGTH ${status_placeholder_length} ALPHABET " " status_placeholder)
    string(LENGTH "${text}" status_text_length)

    # Ensure text part is padded/truncated to placeholder length
    if(status_text_length LESS status_placeholder_length)
      # Pad with spaces if shorter
      string(SUBSTRING "${text}${status_placeholder}" 0 ${status_placeholder_length} status_text_padded)
    elseif(status_text_length GREATER status_placeholder_length)
       # Truncate if longer (or handle differently if needed)
       string(SUBSTRING "${text}" 0 ${status_placeholder_length} status_text_padded)
       message(STATUS "Warning: Status text truncated: ${text}") # Optional warning
    else()
       set(status_text_padded "${text}")
    endif()

    # Handle conditional output (THEN/ELSE)
    if(DEFINED status_then OR DEFINED status_else)
      if(${status_cond}) # Evaluate the condition variable
        string(REPLACE ";" " " status_then "${status_then}") # Join list elements if any
        string(REGEX REPLACE "^[ \t]+" "" status_then "${status_then}") # Trim leading space
        message(STATUS "${status_text_padded} ${status_then}")
      else()
        string(REPLACE ";" " " status_else "${status_else}") # Join list elements if any
        string(REGEX REPLACE "^[ \t]+" "" status_else "${status_else}") # Trim leading space
        message(STATUS "${status_text_padded} ${status_else}")
      endif()
    else()
      # Handle simple value output (no THEN/ELSE)
      string(REPLACE ";" " " status_cond "${status_cond}") # Join list elements if any
      string(REGEX REPLACE "^[ \t]+" "" status_cond "${status_cond}") # Trim leading space
      message(STATUS "${status_text_padded} ${status_cond}")
    endif()
  else()
    # Just print the text if no condition/value provided
    message(STATUS "${text}")
  endif()
endfunction()

# Function merging lists of compiler flags to single string.
# Usage:
#   liquid_merge_flag_lists(out_variable <list1> [<list2>] [<list3>] ...)
function(liquid_merge_flag_lists out_var)
[... 918 lines cut ...]
# programs in the host environment
set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)
```

Some build commands

SSE4
`cmake -DCMAKE_TOOLCHAIN_FILE=(toolchain-location-here) -DBUILD_AUTOTESTS=OFF -DBUILD_BENCHMARKS=OFF -DFIND_SIMD=OFF -DC_SSE4_FOUND=ON -DC_SSE4_FLAGS="-msse4" ..`

AVX
`cmake -DCMAKE_TOOLCHAIN_FILE=(toolchain-location-here) -DBUILD_AUTOTESTS=OFF -DBUILD_BENCHMARKS=OFF -DFIND_SIMD=OFF -D C_SSE4_FOUND=ON -DC_AVX_FOUND=ON -DC_AVX_FLAGS="-mavx" ..`

AVX2
`cmake -DCMAKE_TOOLCHAIN_FILE=(toolchain-location-here) -DBUILD_AUTOTESTS=OFF -DBUILD_BENCHMARKS=OFF -DFIND_SIMD=OFF -DC_SSE4_FOUND=ON -DC_AVX_FOUND=ON -DC_AVX2_FOUND=ON -DC_AVX2_FLAGS="-mavx2" ..`

AVX512
`cmake -DCMAKE_TOOLCHAIN_FILE=(toolchain-location-here) -DBUILD_AUTOTESTS=OFF -DBUILD_BENCHMARKS=OFF -DFIND_SIMD=OFF -DC_SSE4_FOUND=ON -DC_AVX_FOUND=ON -DC_AVX2_FOUND=ON -DC_AVX512_FOUND=ON -DC_AVX512_FLAGS="-mavx512f"`

This appears to all work well.

However, this is **_AI assisted_** with CMake so there might be some errors but this all seems to work. 

Hopefully this helps :)

pclov3r, 2025-04-26:
Actually it appears the proper avx512 build command 

cmake -DCMAKE_TOOLCHAIN_FILE=(toolchain-location-here) -DBUILD_AUTOTESTS=OFF -DBUILD_BENCHMARKS=OFF -DFIND_SIMD=OFF -DC_SSE4_FOUND=ON -DC_AVX_FOUND=ON -DC_AVX2_FOUND=ON -DC_AVX512_FOUND=ON -DC_AVX512_FLAGS="-mavx512f;-mavx512dq;-mavx512vl;-mavx512bw;-mfma" ..

will fail to build and segfauilt trying to run a SSE4 test 

```
-- The C compiler identification is GNU 12.0.0
-- The CXX compiler identification is GNU 12.0.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/x86_64-w64-mingw32-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/x86_64-w64-mingw32-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Skipping   Test C_HAS_SSE4_1:  
-- Skipping   Test C_HAS_SSE4_1: -msse4.2
-- Skipping   Test C_HAS_SSE4_1: /arch:SSE
-- Skipping   Test C_HAS_AVX_1:  
-- Skipping   Test C_HAS_AVX_1: -mavx
-- Skipping   Test C_HAS_AVX_1: /arch:AVX
-- Skipping   Test C_HAS_AVX2_1:  
-- Skipping   Test C_HAS_AVX2_1: -mavx2 -mfma -mf16c
-- Skipping   Test C_HAS_AVX2_1: /arch:AVX2
-- Skipping   Test C_HAS_AVX512_1:  
-- Skipping   Test C_HAS_AVX512_1: -mavx512f -mavx512dq -mavx512vl -mavx512bw -mfma
-- Skipping   Test C_HAS_AVX512_1: /arch:AVX512
-- Preparing  Test C_HAS_NEON_1: ' '
-- Performing Test C_HAS_NEON_1
-- Performing Test C_HAS_NEON_1 - Failed
-- Preparing  Test C_HAS_NEON_2: '-ffast-math'
-- Performing Test C_HAS_NEON_2
-- Performing Test C_HAS_NEON_2 - Failed
-- Preparing  Test C_HAS_NEON_3: '/arch:armv8.0'
-- Performing Test C_HAS_NEON_3
-- Performing Test C_HAS_NEON_3 - Failed
-- Preparing  Test C_HAS_NEON_4: '/arch:armv9.0'
-- Performing Test C_HAS_NEON_4
-- Performing Test C_HAS_NEON_4 - Failed
-- Preparing  Test C_HAS_ALTIVEC_1: ' '
-- Performing Test C_HAS_ALTIVEC_1
-- Performing Test C_HAS_ALTIVEC_1 - Failed
-- Preparing  Test C_HAS_ALTIVEC_2: '-fno-common -faltivec'
-- Performing Test C_HAS_ALTIVEC_2
-- Performing Test C_HAS_ALTIVEC_2 - Failed
-- Preparing  Test C_HAS_ALTIVEC_3: '/arch:altivec'
-- Performing Test C_HAS_ALTIVEC_3
-- Performing Test C_HAS_ALTIVEC_3 - Failed
-- Preparing  Test CXX_HAS_SSE4_1: ' '
-- Performing Test CXX_HAS_SSE4_1
Segmentation fault

```

Will look into later.

pclov3r, 2025-04-26:
Neverminded, It actually works fine!

I cloned a new repo and forgot the update the two files i posted here. Too early in the morning :)

pclov3r, 2025-07-22:
It looks like some of the cleanup to make things easier is good but one issue remains that this PR addresses

https://github.com/jgaeddert/liquid-dsp/pull/411

pclov3r, 2025-08-24:
Also,

```
if (fftw3f_FOUND)
target_link_libraries(${LIBNAME} fftw3f)
endif()
```

needs to be changed to

```
if (fftw3f_FOUND)
target_link_libraries(${LIBNAME} ${fftw3f_LIBRARY})
endif()
```

So it links correctly.