jgaeddert/liquid-dsp #404: Library version detection

szpajder, 2025-04-12:
In liquid-dsp prior to 1.7.0 we had this:

```C
#define LIQUID_VERSION_NUMBER   1006000
```

But now we have this:

```C
#define LIQUID_VERSION_NUMBER ( \
    (uint32_t) ( \
    (LIQUID_VERSION_MAJOR << 16) | \
    (LIQUID_VERSION_MINOR <<  8) | \
    (LIQUID_VERSION_PATCH      )))
```

Which means that `LIQUID_VERSION_NUMBER` for version 1.7.0 equals 67328. That is, it is numerically smaller than any version released prior to it.

Build checks in my app fail if liquid-dsp is not installed or is older than 1.3.0. But now 1.7.0 is older than 1.3.0, so the build fails.

How are we supposed to work with this?

Can you suggest a method of comparing library versions which works **reliably** across all versions?

jgaeddert, 2025-04-12:
Dang, I'm sorry. My original intent for `LIQUID_VERSION_NUMBER` was to compare for version equality between the header and the built object so one could ensure they aren't compiling and linking against two separate versions (see [LIQUID_VALIDATE_LIBVERSION](https://github.com/jgaeddert/liquid-dsp/blob/master/include/liquid.h#L80-L87)). I've been burned by changing behavior before and I'm sorry for the disruption.

A few points:

  1. The reliable way to check for version compatibility is to check ranges for major minor and minor versions separately as suggested in your referenced issue: https://github.com/szpajder/dumphfdl/issues/51#issuecomment-2798488589 however I would probably also ensure that the major version is specifically `1`
  2. I will always maintain major/minor/patch versioning (and not move to something obscure like date-based numbering)
  3. Checking that the (consolidated) version number is greater than a value will certainly break things when version 2.0 is eventually released which will completely change the APIs

Balancing new features has been a real challenge since many changes necessarily involve breaking backwards compatibility, and I would prefer not to increment the major version every few months. Having said that, I know this particular change was avoidable and I apologize for the headache this caused.

szpajder, 2025-04-12:
>     1. The reliable way to check for version compatibility is to check ranges for major minor and minor versions separately as suggested in your referenced issue: [Incompatible with liquid-dsp 1.7.0 szpajder/dumphfdl#51 (comment)](https://github.com/szpajder/dumphfdl/issues/51#issuecomment-2798488589) however I would probably also ensure that the major version is specifically `1`

But `LIQUID_VERSION_MAJOR` and `LIQUID_VERSION_MINOR` don't exist in earlier versions. Hence, this method is unreliable.

Is there any important reason for changing the way `LIQUID_VERSION_NUMBER` is calculated? If you kept it the old way, everything would work as before, for any prior and future version of the library.

jgaeddert, 2025-04-13:
Ah, good point. So here's a simple example of how you can strip the major, minor, patch values from `LIQUID_VERSION_NUMBER`. I realize it's a bit hackey, but it does work given that values prior to version 1.7.0 have a very large version number.

```C
// consistent check for library versions
#include <stdio.h>
#include <stdlib.h>
#include "liquid.h"

int main()
{
    int liquid_version_major = 0;
    int liquid_version_minor = 0;
    int liquid_version_patch = 0;
    if (LIQUID_VERSION_NUMBER >= 1000000) {
        // strip major/minor/patch values using base-10 logic
        liquid_version_major = (LIQUID_VERSION_NUMBER / 1000000);
        liquid_version_minor = (LIQUID_VERSION_NUMBER - 1000000*liquid_version_major) / 1000;
        liquid_version_patch = (LIQUID_VERSION_NUMBER % 1000);
    } else {
        // strip major/minor/patch values using bit masks
        liquid_version_major = (LIQUID_VERSION_NUMBER >> 16) & 0xff;
        liquid_version_minor = (LIQUID_VERSION_NUMBER >>  8) & 0xff;
        liquid_version_patch = (LIQUID_VERSION_NUMBER      ) & 0xff;
        // alterantively we could check for the existence of these macros
        //liquid_version_major = LIQUID_VERSION_MAJOR;
        //liquid_version_minor = LIQUID_VERSION_MINOR;
        //liquid_version_patch = LIQUID_VERSION_PATCH;
    }
    printf("major = %d, minor = %d, patch = %d\n", liquid_version_major, liquid_version_minor, liquid_version_patch);
    return 0;
}
```

closed 2025-04-14 by szpajder