jgaeddert/liquid-dsp #405: Correct scale for spgram get_psd_mag?

brian-armstrong, 2025-04-15:
Hello,

It looks like there's a TODO to set the scale of spgram's get_psd_mag when accumulate is false (forgetting factor > 0). Would this just be the inverse of the sum of the infinite geometric series given by gamma plus some term for alpha? Or if that's not it then I'd be curious what a correct value could be.

Thanks!

jgaeddert, 2025-04-20:
This is a great point! I looked at the code and realized that I set the scale to 0 when accumulate is false (e.g. forgetting factor > 0) which is wrong. The loop filter for integrating the PSD should automatically compensate for the scaling, so when accumulation is true the scaling factor should be 1.

I pushed a change to master to reflect this. I also created a new [example](https://github.com/jgaeddert/liquid-dsp/blob/master/examples/spgramcf_compare_example.c) to compare the difference between "real-time" and "infinite" integration. Here's an example of the output (notice that the scales are roughly equivalent):

[image]

closed 2025-05-16 by jgaeddert

seonlee2013, 2025-05-16:
好啊，邮件收到啦！
Hi, your email is arrived safely