jgaeddert/liquid-dsp #418: Console spammed with errors when using qdsync and underlying qdetector returns tau_hat >= 1

asazernik, 2025-08-20:
```
error [3]: invalid parameter or configuration
  /home/asa/src/ring/3rdparty/liquid-dsp/src/filter/src/firpfb.proto.c:355: firpfb_execute(), filterbank index (325) exceeds maximum (256)
```

Relevant lines are qdsync.proto.c:378 (call to `qdetector_*_get_tau`) and qdsync.proto.c:386 (where it's used to update `pfb_index`). Not sure if the solution is to (as with `tau_hat` values) alias the resulting `pfb_index` into the range [0,`_q->npfb`), or do something else.

jgaeddert, 2025-11-16:
Do you have a simple program you can use to reproduce this? I know you have a PR as a work-around, but I'd like to get a better understanding of why this is happening.

asazernik, 2025-11-23:
Will reduce down to an actual runnable program later, but for a preview here's my general (C++) code.

```cpp
    // constants
    static constexpr float EXCESS_BANDWIDTH_FACTOR = 0.25f;
    static constexpr unsigned FILTER_DELAY_SYMBOLS = 4;
```
...
```cpp
            // initialization: bandwidth on the order of 1e-4 to 1e-2, m_samplesPerSymbol ~= 10
            // which might be wrong! samplesPerSymbol is post-decimation, bandwidth is relative to pre-decimation sample rate
            auto modulationFactor = bandwidth * m_samplesPerSymbol;
            m_synchronizer = qdsync_cccf_create_cpfsk(
                const_cast<uint8_t*>(preambleSeq.data()), preambleSeq.size(),
                LIQUID_FIRFILT_RRC, bitsPerSymbol, modulationFactor, m_samplesPerSymbol,
                FILTER_DELAY_SYMBOLS, EXCESS_BANDWIDTH_FACTOR, LIQUID_CPFSK_SQUARE,
                // null callback/ctx, don't care about the bits
                nullptr, nullptr
            );
```
...
```cpp
qdsync_cccf_set_range(m_synchronizer, 0.3);
```
...
```cpp
        // finelyAdjusted is the output of a nco_crcf downshifter
        qdsync_cccf_reset(m_synchronizer);
        qdsync_cccf_execute(m_synchronizer, finelyAdjusted.data(), finelyAdjusted.size());
        if (qdsync_cccf_is_detected(m_synchronizer))
        {
            return;
        }
```